JChristensen/extEEPROM #1: Read/Writes limited by byte

FranticRain, 2015-04-04:
In your readme, you claim "I/O across block, page and device boundaries is supported." However, when attempting to read from a device such as the ST Micro M24M02-02, you can't even get a full page!

It looks like everything is set up in the code properly to handle this, but using a byte to store length ruins it!

JChristensen, 2015-04-04:
It's odd that I hadn't noticed that. This data logger project ([code](https://github.com/JChristensen/aaLogger_SW), [circuit](https://github.com/JChristensen/aaLogger_HW)) uses two [M24M02 EEPROMs](http://www.mouser.com/Search/ProductDetail.aspx?R=M24M02-DRMN6TPvirtualkey51120000virtualkey511-M24M02-DRMN6TP) and works fine, over a dozen examples having been built by me and others.

Recommend submitting a [SSCCE](http://www.sscce.org/) that demonstrates the alleged issue.

JChristensen, 2015-04-04:
I may have misunderstood. I think you are saying that because the M24M02 has a 256 byte page size, that can't be properly represented when the argument `nBytes` is a byte type, e.g. in `byte write(unsigned long addr, byte *values, byte nBytes)`

FranticRain, 2015-04-04:
You are correct. Thats what I meant. Sorry for being vague.

JChristensen, 2015-04-04:
No worries, good catch, thank you. I think all that is needed is to retype `nBytes` as uint16_t in the `read()` and `write()` functions.

FranticRain, 2015-04-04:
Thats exactly what I did to get it working on my end. Hasn't been tested extensively, but it works for me!

JChristensen, 2015-04-04:
Well that's some testing, so that is worth something! Interesting that I had the variables that track bytes within `read()` and `write()` typed as uint16_t, but somehow the function arguments got overlooked. Thanks again.

FranticRain, 2015-04-04:
No, thank you for making an awesome library that kicks the pants off of Wire.h

JChristensen, 2015-04-04:
Thanks for the feedback, glad you like the library!

commit 215d5ac5 (2015-04-04): Change argument nBytes in write() and read() functions from byte data type to unsigned int. This allows devices with page size of 256 bytes or larger. Fixes #1.
closed 2015-04-04 by JChristensen