JChristensen/extEEPROM #3: Problem with the Example from extEEPROM

jar10623, 2015-06-09:
i open with winRAR

and see this:

void eeRead(uint8_t chunk)
{
    chunk &= 0xFC;                //force chunk to be a multiple of 4
    uint8_t data[chunk];
    uint32_t val = 0, testVal;

    Serial << F("Reading...") << endl;

    uint32_t msStart = millis();
    

it looks like ALL Serial.print in the Example are damage!

i can repair it manually but when does it means 

Serial.print or Serial.println ?

jar10623, 2015-06-09:
line23
        Serial << endl << F("extEEPROM.begin() failed, status = ") << eepStatus << endl;
line27
    Serial << endl << F("Press button to start...") << endl;
line50
    Serial << F("Writing...") << endl;
line54
        if ( (addr &0xFFF) == 0 ) Serial << addr << endl;
line65
    Serial << "Write lapse: " << msLapse << " ms" << endl;

and so on:
line74, 78, 82, 87, 95, 100, 104, 111, 118, 119, 120, 121, 122, 124, 125, 127

JChristensen, 2015-06-09:
https://github.com/JChristensen/extEEPROM/blob/master/examples/eepromTest/eepromTest.ino#L8

closed 2015-06-09 by JChristensen