JChristensen/extEEPROM #5: Compilation error for Arduino Due

mariovaldez, 2015-11-22:
When I tried to compile for the Arduino Due:
```
extEEPROM.cpp:99:7: error: 'TWBR' was not declared in this scope
       TWBR = ( (F_CPU / twiFreq) - 16) / 2;
```

It compiles if I check if we are on an AVR:
```
    #ifdef __AVR__
      TWBR = ( (F_CPU / twiFreq) - 16) / 2;
    #endif
```
However then I think the I2C speed would default to 100kHz (I'm not sure, I'm just starting to use the Arduino Due).

PaoloP74, 2015-12-05:
Test my fork --> https://github.com/PaoloP74/extEEPROM

searobin, 2016-11-02:
Hi.
The released library within Arduino IDE is not working anymore.
This because the Wire.h does not include anymore the BUFFER_LENGTH definition.

The owner of the library should release an update.

Please let me know

Ciao
Severino

jzalger, 2021-06-02:
> Test my fork --> https://github.com/PaoloP74/extEEPROM

Nailed it. Thanks 👍

JChristensen, 2022-01-15:
https://github.com/JChristensen/JC_EEPROM now uses setClock() instead of direct register manipulation.

closed 2022-01-15 by JChristensen