JChristensen/extEEPROM #9: New fork fails!

alanoatwork, 2017-06-27:
Not sure where to vent but Paolo74's fork of your library just showed up in the Arduino IDE as version 3.3.3. My code no longer compiles because the initialization fails to find the enumeration:
  `ee.begin(twiClock100kHz);`
results in:
```
exit status 1
'twiClock100kHz' was not declared in this scope

```
Paolo74's fork doesn't allow issues to be raised so I thought I would raise it here. Hope you can forward this on or help explain why the newer library is failing.

JChristensen, 2017-06-28:
Hello, this is the first I have heard of this issue. I do not know PaoloP74 but I left a note on his latest commit, https://github.com/PaoloP74/extEEPROM/commit/06ea701d077e5bcd57a920586b1dd35e3056b6fd

alanoatwork, 2017-06-28:
Thanks. Not sure why I can raise issues on his fork but then again I’m not a GitHub power user.

Regards,

Alan

PaoloP74, 2017-06-28:
Hi @alanoatwork , I corrected the reported error.
You can try the new version here: https://github.com/PaoloP74/extEEPROM/releases/tag/v3.3.4
It will be available on the library manager from Saturday.

Thanks also for @JChristensen for reporting.
Sorry for my horrible english.

alanoatwork, 2017-06-29:
Nope, still have the same issue with both 3.3.3 and 3.3.4.

PaoloP74, 2017-06-29:
Use
`ee.begin(ee.twiClock100kHz);`

alanoatwork, 2017-06-29:
Works much better!

Thanks,

Alan

closed 2022-01-15 by JChristensen