JChristensen/extEEPROM #10: stm32f103's Wire.h have no setClock currently

lefedor, 2017-07-06:
I've hotfixed it this way currently:

extEEPROM.cpp:

byte extEEPROM::begin(twiClockFreq_t twiFreq)
{
    Wire.begin();
    #ifndef _LIBMAPLE_STM32_H_
		Wire.setClock(twiFreq);
    #endif
    Wire.beginTransmission(_eepromAddr);
    if (_nAddrBytes == 2) Wire.write((byte)0);      //high addr byte
    Wire.write((byte)0);                            //low addr byte
    return Wire.endTransmission();
}

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/Wire

victorpv, 2017-09-10:
You don't need that fix anymore, the stm32 wire class now has setClock.

JChristensen, 2022-01-15:
See https://github.com/JChristensen/JC_EEPROM

closed 2022-01-15 by JChristensen