JChristensen/extEEPROM #24: Reset function

JosepMir, 2020-03-15:
http://ww1.microchip.com/downloads/en/AppNotes/01028a.pdf describes a way to  Reset the 24LCxxx EEPROM.  Any chance this library could do it? 
Even using the underling Wire.cpp or the native twi.cpp seems not enough low level to transmit "9 bits". (i can only see TWI_WriteByte)

[image]

JChristensen, 2020-03-19:
Greetings, I was not aware of this reset sequence. Do you have an actual need for it?

JosepMir, 2020-03-19:
:D yes, some times my EEPROM would hang up. Right after startup, while doing eep.begin(..) , the endTransmission() returned !=0.
So then I found this procedure. Not sure if it's standard or only some vendors.

Meanwhile I modified my library locally, too specific to my case for a pull request,,,
`
void extEEPROM::resetAndBegin()
{
	Wire1.end();
	
	delay(5); //just in case
	
	  pinMode(SDA1, OUTPUT); 
	  pinMode(SCL1, OUTPUT); 
	  
	  //TODO: la rutina sha de fer cada vegada q endtransmission doni wrong status! dins extEEPROM p ex
	  /* Send START condition */
	  //A HIGH to LOW transition on the SDA line while SCL is HIGH defines a START condition.
	  SCL(HIGH);
	  SDA_HIGH();
	  SDA_LOW();
	  SCL(LOW);
	  

	  SDA_HIGH();
	  //digitalWrite(SDA, HIGH); //NO. because " while allowing the SDA line to float,"  https://www.microchip.com/forums/m898899.aspx
	  //send 9 clock pulses to reset slaves
	  for(int i = 0; i < 9; i++){
		  SCL(HIGH);
		  SCL(LOW);
	  }
	  
	  /* Send START condition */
	  //A HIGH to LOW transition on the SDA line while SCL is HIGH defines a START condition.
	  SCL(HIGH);
	  SDA_HIGH();
	  SDA_LOW();
	  SCL(LOW);
	  
	  /* Send STOP condition */
	  //A LOW to HIGH transition on the SDA line while SCL is HIGH defines a STOP condition.
	  SCL(HIGH);
	  SDA_LOW();
	  SDA_HIGH();
	  SCL(LOW);
	  
	  
	  //TODO: reactivate Wire.begin
	  
	  delay(5); //just in case
	  
	  return begin(twiClock400kHz, &Wire1);
}

void extEEPROM::SDA_LOW(){
	pinMode(SDA1, OUTPUT);
	digitalWrite(SDA1, LOW);
	delayMicroseconds(5);
}

//we dont push explicitely high, we let the pin float and the pullup will set it HIGH
void extEEPROM::SDA_HIGH(){
	pinMode(SDA1, INPUT);
	delayMicroseconds(5);
}

 

//TODO: same for SCL?! not setting it up but letting it float
void extEEPROM::SCL(uint32_t dwVal){
	digitalWrite(SCL1, dwVal);
	delayMicroseconds(5);
}

byte extEEPROM::safeWrite(unsigned long addr, byte *values, unsigned int nBytes)
{
	 uint8_t txStatus =0;
	 int trials=0;
	 
	  txStatus = write(addr, values, nBytes);
	  if (txStatus==0) {
		  return txStatus;
	  }
	  
	  while (txStatus != 0 && trials <5) {
		 delay(100);
		 resetAndBegin();
		 txStatus = write(addr, values, nBytes);
		 trials++;
		 
	 } 
    return txStatus;
}
`

JChristensen, 2020-03-19:
At least for the AVR microcontrollers (e.g. ATmega328P) the actual hardware is limited to working with 8-bit bytes as near as I can tell. I suppose this could vary depending on the hardware. Then there is always bit-banging but that's not a hoop I care to jump through. According to the application note, bus noise or unexpected resets could cause an EEPROM to be in an unknown state. Those seem mostly like design issues so I might look to address those first. Another option may be controlling the EEPROM's power from the MCU.

JChristensen, 2020-03-19:
> :D yes, one time my EEPROM hanged up. Right after startup, while doing eep.begin(..) , the endTransmission() returned !=0.

And simply retrying the transmission did not work?

closed 2020-03-19 by JChristensen

JosepMir, 2020-03-19:
retrying was not there in my code. I turned off/on the whole PCB, and still the same issue. Maybe a 100nF capacitor was holding the EEPROM alive....

JosepMir, 2020-03-19:
thank you anyway ;D