JChristensen/extEEPROM #27: Suggestion for write function

amazed-hub, 2021-12-13:
Hello,

Wouldn't it be better if delay was moved after the conditional break?
This way 500us would not be wasted if eeprom write is already complete at the last iteration.

```
//wait up to 50ms for the write to complete
for (uint8_t i=100; i; --i) {
    //delayMicroseconds(500);                     //no point in waiting too fast
    Wire.beginTransmission(ctrlByte);
    if (_nAddrBytes == 2) Wire.write(0);        //high addr byte
    Wire.write(0);                              //low addr byte
    txStatus = Wire.endTransmission();
    if (txStatus == 0) break;
    **delayMicroseconds(500);                     //new place for delay**
}
```

Best regards,
amazed

JChristensen, 2021-12-13:
Datasheet says max write cycle time is 5ms. Do many writes complete in less than 1/10 of that time?

closed 2021-12-13 by amazed-hub

amazed-hub, 2021-12-13:
You're right, my suggestion wouldn't make any difference, I misled myself.