JChristensen/extEEPROM #29: Writing structure in pages...

AlexArGC, 2022-01-10:
Hello dear developer. You have got a very good library for working with an external EEPROM. Can you make an example of how to save any structure in the EEPROM no more than a page?

JChristensen, 2022-01-10:
Thanks for the kind words.
When you say "any structure," you don't mean a `struct`, but rather a byte array like `myData`?
The & character is not needed since `myData` is an array.
Try this instead:
```c++
byte i2cStat = myEEPROM.write(42, myData, 10);
```

AlexArGC, 2022-01-10:
I mean structures. Everything is clear with bytes. But for example, the EEPROM.h library Writing the entire structure to the flash can be done with one command

`EEPROM.put (EEPROM_START_ADR, Wither);`

Where Wither is a structure like:
```
struct WitherStruct {
  double KpT;                           // 4 byte
  double KiT;                           // 4 byte
  double KdT;                           // 4 byte
  double KpH;                           // 4 byte
  double KiH;                           // 4 byte
  double KdH;                           // 4 byte
  byte Rezhim;                          // 1 byte
  double TempU;                         // 4 byte
  int TempD;                            // 2 byte
  double HudU;                          // 4 byte
  int HudD;                             // 2 byte
  long TimeU;                           // 4 byte
  long long TimeStart;                  // 8 byte

  long long TimeStamp;                  // 8 byte
  long gmtOffset;                       // 4 byte

  byte flPumpCold;                      // 1 byte
  byte flPumpHud;                       // 1 byte
  byte flLight;                         // 1 byte
  byte flIrWarm;                        // 1 byte

  byte flAirCold;                       // 1 byte
  byte flAirHud;                        // 1 byte
  byte flAirOut;                        // 1 byte
  byte flAirDoor;                       // 1 byte
  bool flOnLine;                        // 1 byte
  bool flUpdateFW;                      // 1 byte
  bool flUpdateOverInternetFW;          // 1 byte
};                                      // summ 72 byte
WitherStruct Wither;                    //
```

It is very comfortable. The function will itself determine the size of the structure or array and write it to memory.

AlexArGC, 2022-01-10:
> The & character is not needed since `myData` is an array.

Thanks, I guessed. Then you probably need to fix the example :-)

JChristensen, 2022-01-10:
> > The & character is not needed since `myData` is an array.
> 
> Thanks, I guessed. Then you probably need to fix the example :-)

I didn't know where that example was from. I see it's in the ReadMe file, which has other issues too. Consider it done and thanks for finding it.

JChristensen, 2022-01-10:
This may be above my pay grade. How shall we write a function such as `put(addr, struct)`?
The function can only determine the size of `struct` if it either (A) Has the definition of `struct`, or (B) Has a pointer to `struct` and its size, e.g. `put(addr, struct*, size)`
> I mean structures. Everything is clear with bytes. But for example, the EEPROM.h library Writing the entire structure to the flash can be done with one command
> 
> `EEPROM.put (EEPROM_START_ADR, Wither);`
> 
> Where Wither is a structure like:
> 
> ```
> struct WitherStruct {
>   double KpT;                           // 4 byte
>   double KiT;                           // 4 byte
> ...
>   bool flUpdateOverInternetFW;          // 1 byte
> };                                      // summ 72 byte
> WitherStruct Wither;                    //
> ```
> 
> It is very comfortable. The function will itself determine the size of the structure or array and write it to memory.

JChristensen, 2022-01-11:
What do you think of this approach?
```c++
#include <extEEPROM.h>      // https://github.com/JChristensen/JC_EEPROM
#include <Streaming.h>      // https://github.com/janelia-arduino/Streaming

struct myStruct {
    int       a;
    long      b;
    float     c;
    double    d;
    byte      e;
    bool      f;
};

extEEPROM eep(kbits_256, 1, 64); // 24LC256

void setup()
{
    Serial.begin(115200);
    Serial << F( "\n" __FILE__ "\n" __DATE__ " " __TIME__ "\n" );
    eep.begin(extEEPROM::twiClock400kHz);

    myStruct w { 1, 2, 3, 4, 5, true };
    printStruct(w);

    Serial << "Writing\n";
    eep.write(0, reinterpret_cast<byte*>(&w), sizeof(w));

    Serial << "Reading\n";
    myStruct r;
    eep.read(0, reinterpret_cast<byte*>(&r), sizeof(r));
    printStruct(r);
}

void loop() {}

void printStruct(myStruct s)
{
    Serial << s.a << ' ' << s.b << ' ' << s.c << ' ' << s.d << ' ';
    Serial << s.e << ' ' << s.f << endl;
}
```

JChristensen, 2022-01-15:
Nothing heard; closing issue.

closed 2022-01-15 by JChristensen